tomeshnet/prototype-cjdns-pi #397: NPM updating

makeworld-the-better-one, 2019-04-27:
```

   ╭───────────────────────────────────────────────────────────────╮
   │                                                               │
   │       New minor version of npm available! 6.4.1 → 6.9.0       │
   │   Changelog: https://github.com/npm/cli/releases/tag/v6.9.0   │
   │               Run npm install -g npm to update!               │
   │                                                               │
   ╰───────────────────────────────────────────────────────────────╯
```

Got this message when installing. Should we run `npm install -g npm` whenever we install NodeJS or use npm for modules?

darkdrgn2k, 2019-06-16:
Do you want to update every minor version without testing?

makeworld-the-better-one, 2019-06-19:
No, I just wasn't sure whether it should be done or not, I've never seen it before on the nodes. I'm fine to keep it at the version it is now, and I know we're updating NodeJS in develop.

closed 2019-06-19 by makeworld-the-better-one