tomeshnet/prototype-cjdns-pi #398: Patchfoo fix

makeworld-the-better-one, 2019-04-27:
As I noticed [here](https://github.com/tomeshnet/prototype-cjdns-pi/pull/392#issuecomment-487296572), Patchfoo can no longer be downloaded from Github, and shouldn't be either, even though there is a commit history.

Previously, Patchfoo would be downloaded from Github and refs to git-ssb would be removed, so that difficulty of installing that wouldn't be needed. Looking at the [newer version of Patchfoo](https://git.scuttlebot.io/%25YAg1hicat%2B2GELjE2QJzDwlAWcx0ML%2B1sXEdsWwvdt8%3D.sha256), it uses git-ssb a lot for displaying those kinds of messages, and that part of it can't simply be removed. For example, here's all the times `git` is mentioned in the Patchfoo source:
```
package.json
35:    "ssb-git": "^1.1.0",
48:  "homepage": "https://git.scuttlebot.io/%25YAg1hicat%2B2GELjE2QJzDwlAWcx0ML%2B1sXEdsWwvdt8%3D.sha256",
50:    "type": "git",

README.md
15:- Render core ssb message types, git-ssb message types, and raw messages.
23:Find [this guide](%VaSj08AbdhIa4itK4z8Z91G80o2h5OhRLCEEO6MhAcU=.sha256) [on github](https://github.com/noffle/sailing-patchfoo) or [on
24:SSB](http://git.scuttlebot.io/%25VaSj08AbdhIa4itK4z8Z91G80o2h5OhRLCEEO6MhAcU%3D.sha256).
30:- [git-ssb][]
43:git clone ssb://%YAg1hicat+2GELjE2QJzDwlAWcx0ML+1sXEdsWwvdt8=.sha256 patchfoo
53:git clone ssb://%YAg1hicat+2GELjE2QJzDwlAWcx0ML+1sXEdsWwvdt8=.sha256 patchfoo
144:[git-ssb]: %n92DiQh7ietE+R+X/I403LQoyf2DtR3WQfCkDKlheQU=.sha256

lib/render.js
406:Render.prototype.gitCommitBody = function (body) {

doc/install-guide.html
5:<title>ssb-npm + git-ssb + patchfoo install</title>
13:<h2>Installing ssb-npm + git-ssb + patchfoo</h2>
44:<h3>Install git-ssb</h3>
46:ssb-npm install -g git-ssb@2.6.1 \
54:git clone ssb://%YAg1hicat+2GELjE2QJzDwlAWcx0ML+1sXEdsWwvdt8=.sha256 patchfoo
64:git clone ssb://%YAg1hicat+2GELjE2QJzDwlAWcx0ML+1sXEdsWwvdt8=.sha256 patchfoo

lib/app.js
15:var Git = require('ssb-git')
88:  this.git = new Git(this.sbot, this.config)
1198:  // get line comments for a git-update message and git object id.
1200:  // but we have message id and git object hash.
1201:  // look up the git object hash for each line-comment
1202:  // to verify that it is for the git object file we want
1235:      self.git.getObjectAtPath({
1340:  var tmpPath = path.join(os.tmpdir(), '.git_vtag_tmp' + Math.random().toString('36'))
1341:  // use a temp file to work around https://github.com/nodejs/node/issues/13542
1351:    self.git.extractSignature(obj, function (err, parts) {

lib/render-msg.js
272:    case 'git-repo': return this.gitRepo(cb)
273:    case 'git-update': return this.gitUpdate(cb)
274:    case 'pull-request': return this.gitPullRequest(cb)
833:RenderMsg.prototype.gitRepo = function (cb) {
842:      'git clone ',
853:RenderMsg.prototype.gitUpdate = function (cb) {
866:      self.wrap(h('div.ssb-git-update',
867:        'git push ', a, ' ',
872:          var path = id && ('/git/' + type + '/' + encodeURIComponent(id)
883:          var path = '/git/commit/' + encodeURIComponent(commit.sha1)
888:            self.render.gitCommitBody(commit.body)
893:          var path = '/git/tag/' + encodeURIComponent(tag.sha1)
902:            tag.body ? self.render.gitCommitBody(tag.body) : ''
915:RenderMsg.prototype.gitPullRequest = function (cb) {
1357:function expandDigitToSpaces(n) {
1362:  return line.replace(/\d/g, expandDigitToSpaces).split('')
1891:          href: self.toUrl('/git/commit/' + self.c.commitId + '?msg=' + encodeURIComponent(self.c.updateId))
1894:          href: self.toUrl('/git/line-comment/' +

lib/serve.js
453:    case '/git': return this.git(m[2])
1987:Serve.prototype.git = function (url) {
1990:    case 'object': return this.gitObject(m[2])
1991:    case 'commit': return this.gitCommit(m[2])
1992:    case 'tag': return this.gitTag(m[2])
1993:    case 'tree': return this.gitTree(m[2])
1994:    case 'blob': return this.gitBlob(m[2])
1995:    case 'raw': return this.gitRaw(m[2])
[... 91 lines cut ...]
2579:    self.wrapPage('git diff'),
2586:  self.app.git.getObjectMsg({
2597:      self.wrapPage('git diff ' + revs),
2605:    var blob1Url = '/git/blob/' + rev1 +
2608:    var blob2Url = '/git/blob/' + rev2 +
2636:          self.app.git.openObject({
2640:          self.app.git.openObject({
2645:          self.app.git.guessCommitAndPath({
2656:            pull.collect(done())(self.app.git.readObject(obj1))
2657:            pull.collect(done())(self.app.git.readObject(obj2))
2664:              cb(null, self.gitDiffTable(diff, lineComments, {
2674:      self.wrapPage('git diff'),
2680:Serve.prototype.gitDiffTable = function (diff, lineComments, lineCommentInfo) {
2794:Serve.prototype.gitLineComment = function (path) {
2818:    self.app.git.diffFile({
2831:        path = '/git/blob/' + file.hash[1]
2837:        path = '/git/diff/' + file.hash[0] + '..' + file.hash[1]
2856:Serve.prototype.gitObjectLinks = function (headMsgId, type) {
2859:    self.app.git.getObjectMsg({
2870:      var path = '/git/' + type + '/' + id
```

So git-ssb will need to be installed. To install [git-ssb](https://git.scuttlebot.io/%25n92DiQh7ietE%2BR%2BX%2FI403LQoyf2DtR3WQfCkDKlheQU%3D.sha256), we also need to install [ssb-npm](https://git.scuttlebot.io/%25iqhz%2FsQCZCSp91JYAqfQPzHuDYrjw1geKPf1wJ1CvlA%3D.sha256), and [ssb-npm-registry](https://git.scuttlebot.io/%25pFqjcdVKHqsrtOjVEAVZeCF0iY4s%2B3Hr0vA3EFCq5UM%3D.sha256) (also required directy by Patchfoo). All these installations need a connection to the SSB network during the installation, meaning that for us to install Patchfoo like this, ssb-server will already need to be installed and peered with a pub or two, and successfully synced everything.

This feels like a huge mess and a lot of work, so I made this issue before diving in. One solution might be to host the required blobs ourselves, so that an SSB network connection isn't required. But these blobs update and change all the time, and keeping track of them won't work very well and will just add another job for us.

makeworld-the-better-one, 2019-04-27:
At the moment I would suggest a removal of the module, because it does not install at all right now. I am not in favour of using a very old version on Github instead.

@darkdrgn2k Thoughts on everything above?

darkdrgn2k, 2019-05-05:
Upstream patch no in PR to install correct Patchfoo install #400  not sure if it will work with version bump of Node.js

If you want to test it try  `git cherry-pick 107fae35f23363da38f7c5d0feadb5b0d9010190` before install

makeworld-the-better-one, 2019-05-18:
This has been hacked around because of #400, but still needs a real fix and Patchfoo update in the long run.

darkdrgn2k, 2019-05-25:
> but still needs a real fix and Patchfoo update in the long run.

don't hold your breath
this is the same issue we had with git-ssb from the beggining.

closed 2019-07-08 by darkdrgn2k