tomeshnet/prototype-cjdns-pi #401: IPFS dnsaddr peering

darkdrgn2k, 2019-05-05:
IPFS vv.20 added  supports `/dnsaddr/bootstrap.libp2p.io` like peering that does not require an ID
(changelog)[https://blog.ipfs.io/83-go-ipfs-0-4-20/#changed-ipfs-swarm-connect]

We maybe be able to use this  to stop relaying on curl and nodeinfo.json providing a more universal peering method over CJDNS

https://github.com/tomeshnet/prototype-cjdns-pi/blob/master/scripts/ipfs/ipfs-swarm.sh#L19-L26

Reason revealing IDs or using this method is possibly risky
Question was "What attack vector could an  impersonated node have?"
@aschmahmann said
>there are a number of possible issues that may/may not be relevant for your case. For example, if I know that I can make sure you bootstrap from me then I can give you lots of nodes that are controlled by me. This then gives me the ability to make you think that you've made data available to the IPFS network at large, but in reality I've censored out anything I feel like.

makeworld-the-better-one, 2019-05-13:
This requires DNS. I'm not sure what you're saying about making it easier. Could you explain further?

darkdrgn2k, 2019-06-18:
It SUPPORST dns but does not REQUIRE dns.

The reason for this is when you have a cluster of ipfs nodes and you put them all on the same DNS  hostname you dont know WHICH ipfs server you will get (and as such which ipfs public key to use) so dnsaddr does NOT require a public key

meaning we dont need node-info anymore

darkdrgn2k, 2019-07-01:
Turns out currently it requires a DNS name
(hosts dont even work)

Will not work for us :(

closed 2019-07-31 by makeworld-the-better-one