tomeshnet/prototype-cjdns-pi #421: Running ipfs swarm connect on both nodes

makeworld-the-better-one, 2019-06-04:
As I found out [here](https://github.com/ipfs/go-ipfs/issues/6271#issuecomment-498839894), you need to run `ipfs swarm connect` on both nodes to actually successfully establish a priority connection. We [do this](https://github.com/tomeshnet/prototype-cjdns-pi/blob/develop/scripts/ipfs/ipfs-swarm.sh#L36) in our code with direct mesh peers, but only on [startup of the IPFS service](https://github.com/tomeshnet/prototype-cjdns-pi/blob/develop/scripts/ipfs/ipfs.service#L10). 

This means that sometimes this command will only go one way (when a new node joins), or that sometimes the connection is joined much later (when a node sends it out to a node that doesn't have IPFS started, and then that node later starts it and sends it back). I am checking [here](https://github.com/ipfs/go-ipfs/pull/6373#issuecomment-498859272) and previously [here](https://github.com/ipfs/go-ipfs/issues/6271#issuecomment-498857658) how large the window is between call and response, but that still doesn't solve the former problem about when only one node issues the command.

One solution would be to turn `ipfs-swarm.sh` into a loop, every 30 seconds or whatever the window is. It could also potentially keep track of which nodes it had establish priority connections with, and only the contact the ones it hadn't.

darkdrgn2k, 2019-07-19:
I'd rather use some sort of discovery protocol possibly like #385

makeworld-the-better-one, 2019-08-06:
I'm still confused about how this plays out, I've asked to confirm [here](https://github.com/ipfs/go-ipfs/issues/6271#issuecomment-518824182).