tomeshnet/prototype-cjdns-pi #452: Hostapd Automatic Channel Selection

leosan902, 2019-10-17:
Where is the config file to change this configuration?
CONFIG_ACS=y
That is needed to enable Automatic Channel Selection,
documentaion:
https://gitlab.gwdg.de/eap-saml/hostapd/blob/master/src/ap/acs.c
https://wireless.wiki.kernel.org/en/users/documentation/acs

darkdrgn2k, 2019-10-25:
Seems to be a HOSTAPD thing

```
# If CONFIG_ACS build option is enabled, the channel can be selected
# automatically at run time by setting channel=acs_survey or channel=0, both of
# which will enable the ACS survey based algorithm.
```

CONFIG_ACS is a BUILD OPTION which means it needs to be part of the parameters when compiled.  Im not sure what debian/ubuntu use for their flags when building their packages.

If they do not have it enabled, then you must re-compile hostapd from source and make sure the flag is set.

leosan902, 2019-10-25:
And my alfa network that i'm using as a AP doesnt have this drivers
ath5k
ath9k 
ath10k

darkdrgn2k, 2019-10-25:
Not sure what board and distro your running this on.  Or which wifi card you are running.

note: out-of-the-box prototype does not have a way of using anything but the onboard wifi for access point (although can be manually hacked into place)

closed 2019-11-04 by leosan902