tomeshnet/prototype-cjdns-pi #473: SSB layer, change Patchfoo for Oasis

zicmama, 2020-02-17:
I have just installed "prototype-cjdns-pi" on a Rpi3B+
I still need to dig into code & behaviour....

Maybe I'll try to add a few more libre software to your install procédure...
One I'd like to add is https://github.com/fraction/oasis/

Do you have special documentation I should read?
Tanks

benhylau, 2020-02-17:
You will see each module is [in its own folder](https://github.com/tomeshnet/prototype-cjdns-pi/tree/master/scripts), so I imagine Oasis to be another. You can look at [the IPFS one](https://github.com/tomeshnet/prototype-cjdns-pi/tree/master/scripts) as an example, where there is an executable `install` and `uninstall` file, then [install2](https://github.com/tomeshnet/prototype-cjdns-pi/blob/master/scripts/install2#L457-L460) decides whether to run it.

This format makes it easy for @darkdrgn2k's parallel effort to package these modules into debian packages for rpi and other debian-based systems.

darkdrgn2k, 2020-02-22:
Looks like a very nice interface!

>  No browser JavaScript! Oasis has strict security rules that prevent any JavaScript from running in your browser, which helps us make Oasis accessible and easy to improve.

Very interesting!