tomeshnet/prototype-cjdns-pi #474: Raspbian buster (RPi3B+) npm: command not found

zicmama, 2020-02-17:
While installing

cat /etc/os-release 
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
With this options (Yggrasil = NO)
[Does yggdrasil really decentralised? ](https://discuss.ipfs.io/t/dht-and-p2p-swarm-peers-discovery-yggdrasil-mesh-networks/6549/7)
```
Enter the name of your mesh network (default: tomesh): tomesh
Install Yggdrasil routing engine [Y/n]? n
Yggdrasil routing engine will be skipped
Install Mesh Point Interface [Y/n]? n
Mesh Point Interface will be skipped
Install Ad-Hoc Interface [Y/n]? n
Ad-Hoc Interface will be skipped
Install WiFi Access Point [Y/n]? n
WiFi Access Point will be skipped
Install Basic Firewall [Y/n]? 

Basic Firewall will be enabled
Install Internet Gateway over CJDNS [Y/n]? 

Internet Gateway over CJDNS will be enabled
Install IPFS [Y/n]? 

IPFS will be enabled
Install IPFS Pi Stream [y/N]? y
IPFS Pi Stream will be enabled
Install SSB [Y/n]? 

SSB will be enabled
Install SSB PatchFoo Web Client [Y/n]? 

SSB PatchFoo Web Client will be enabled
Install SSB Web Pi Client (Experimental) [y/N]? 

SSB Web Pi Client (Experimental) will be skipped
Install Yrd (a cjdns command-line tool) [Y/n]? 

Yrd (a cjdns command-line tool) will be enabled
Install Prometheus Node Exporter [Y/n]? 

Prometheus Node Exporter will be enabled
Install Prometheus Server [Y/n]? 

Prometheus Server will be enabled
Install Grafana [Y/n]? 

Grafana will be enabled
Install Hyperboria Domain Name Servers [Y/n]? 

Hyperboria Domain Name Servers will be enabled
Install Hyperboria Network Time Protocol Server [Y/n]? 

Hyperboria Network Time Protocol Server will be enabled
Install Non-Essential Tools Useful for Network Analysis [Y/n]? 

Non-Essential Tools Useful for Network Analysis will be enabled
Install Hardware Watchdog [Y/n]? 

Hardware Watchdog will be enabled
Reading package lists... Done
```
Answered "Compiling cjdns from source"

But finished with !?
```
sudo: npm: command not found
```

Must be related to this Warning?

```
================================================================================
================================================================================

                              DEPRECATION WARNING                            

  Node.js 8.x LTS Carbon is no longer actively supported!

  You will not receive security or critical stability updates for this version.

  You should migrate to a supported version of Node.js as soon as possible.
  Use the installation script that corresponds to the version of Node.js you
  wish to install. e.g.

   * https://deb.nodesource.com/setup_10.x — Node.js 10 LTS "Dubnium" (recommended)
   * https://deb.nodesource.com/setup_12.x — Node.js 12 LTS "Erbium"

  Please see https://github.com/nodejs/Release for details about which
  version may be appropriate for you.

  The NodeSource Node.js distributions repository contains
  information both about supported versions of Node.js and supported Linux
  distributions. To learn more about usage, see the repository:
    https://github.com/nodesource/distributions

================================================================================
================================================================================

Continuing in 20 seconds ...

## Installing the NodeSource Node.js 8.x LTS Carbon repo...
```

What is your advice to overcome this issue?

darkdrgn2k, 2020-02-22:
Try running this with Develop branch.  It has many updates including node.js version bump and better support for debian 10.

darkdrgn2k, 2020-03-29:
We have now pushed v0.5RC1 to master.  This should solve allot of problems experiencing with the aging versions in 0.4

closed 2020-03-29 by darkdrgn2k