tomeshnet/prototype-cjdns-pi #478: Yggdrasil & cjdns optional install

zicmama, 2020-02-25:
In most condition, as I am using ipfs and ssb layers there is no need for meto have that low level layers... Making me mostly confusing.

Could it be possible to activate only storage and ssb + application layer?
WAN & LAN works like a charm using ssb Pubs interconnecting ipfs swarms

darkdrgn2k, 2020-03-29:
@zicmama - Sorry for the delay in responding to you about this.

We just pushed V0.5RC1 to master with many improvements. I'm not sure if you can install the stack without CJDNS or  Yggdrasil since there are places it looks for it during the install to make the install work. We have done work to allow one OR the other to be installed
(for one the node name is derived from these two see https://github.com/tomeshnet/prototype-cjdns-pi/blob/master/scripts/shared/hostname/install )

However the stack  is not depended on this to actually work. You can simply disable them by running 'sudo systemctl disable yggdrasil' (reboot required)

As for Firewall ,CJDNS/YGGDARSIL themselves do not apply any firewall rules.  There is a standard list of rules we created to make sure nodes installed are not "default open" to the world especially since Yggdrasil issues Yggdrasil addressable ipv6 addresses to AP clients.

Firewall is optional when using CUSTOM install, and can disable by running `sudo systemctl disable netfilter-persistent` or simply deleting the rules in `/etc/iptables/` (reboot needed for both)
(reference https://github.com/tomeshnet/prototype-cjdns-pi/blob/master/scripts/firewall/install#L19-L20 )

You can also add additional rules into the correct file to open the firewall holes. If there is something we missed id like to hear it so we can add it to our current default firewall.

Moving forward we recognize that this stack has been used for much more then a playground for d-web things which it was originally envisioned for. This is why we are working towards building a "mesh repository" for Debian which will have  DEB packages that can be used to build a node YOUR WAY.

Its still in its infancy but a proof of concept repo can be found here https://github.com/darkdrgn2k/packages

This repo automatically generates DEB packages for multiple platforms and uploads them here
http://meshwithme.online/deb/repos/apt/debian/

(sadly i have not found a way to cross compile SSB yet into ARM so it currently only has the AMD64 version)

makeworld-the-better-one, 2020-04-11:
If you want to follow the issue about SSB on ARM, it's [here](https://github.com/darkdrgn2k/packages/issues/11).