tomeshnet/prototype-cjdns-pi #491: Missing Artifact files

Toskin, 2022-06-15:
Files downloaded from gh.circle-artifacts.com seems to be missing from said site. Attempts to directly access files through browser drops to empty page with missing file notice. Issue happens for both CJDNS and Yggdrasil

Paste from installation process: 

--2022-06-15 08:35:08--  https://2375-115685026-gh.circle-artifacts.com/0/yggdrasil-0.3.12-armhf.deb
Resolving 2375-115685026-gh.circle-artifacts.com (2375-115685026-gh.circle-artifacts.com)... 52.86.102.25, 35.168.223.154, 100.25.6.230, ...
Connecting to 2375-115685026-gh.circle-artifacts.com (2375-115685026-gh.circle-artifacts.com)|52.86.102.25|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-06-15 08:35:08 ERROR 404: Not Found.

Program is being installed on Pi 3B.

darkdrgn2k, 2022-08-18:
Yes main has been broken for some time, this is one of the issue (ygg changed the location of their files)

Develop is in a more usable state, i think we should merge develop->master

darkdrgn2k, 2022-12-26:
Fixed and merged from dev to master sooj

closed 2022-12-26 by darkdrgn2k